kike-canaries/canairio_sensorlib #9: BATT charge level appears every second instead sample time via serial

danielbernalb, 2020-10-06:
Serial output:
_-->[BATT] chargelevel: 0%
-->[HPMA] read > done! PM1:000 PM25:000 PM10:000 H:00% T:00°C
-->[BATT] chargelevel: 0%
-->[BATT] chargelevel: 0%
-->[BATT] chargelevel: 0%
-->[BATT] chargelevel: 0%
-->[BATT] chargelevel: 0%
-->[HPMA] read > done! PM1:000 PM25:000 PM10:000 H:00% T:00°C
-->[BATT] chargelevel: 0%
-->[BATT] chargelevel: 0%
-->[BATT] chargelevel: 0%
-->[BATT] chargelevel: 0%
-->[BATT] chargelevel: 0%
-->[HPMA] read > done! PM1:000 PM25:000 PM10:000 H:00% T:00°C_
-->[BATT] chargelevel: 0%

hpsaturn, 2020-10-06:
Thanks for report it but this issue is from `canairio_firmware` repo, the battery level for now, is handled by the main firmware, the sensorlib can't using the battery.

hpsaturn, 2020-11-03:
I'm going to close this issue because is for `canairio_firmware` repo. Please report it in that repo.

closed 2020-11-03 by hpsaturn