kike-canaries/canairio_sensorlib #69: Sleep mode  and autoclean for sensirion sps30

roberbike, 2021-04-25:
For the sensirion sps30, add new functions to achieve the following improvements:

**Sleep mode to save battery**. 2 18650 1.5Abatteries , only can run for 6 hours aprox.

**Autoclean** activated for fixed stations. By default autoclean happens 1 time per week, but if the station is power off, the counter goes to 0. There is a function to make a clean on start up, in order to keep the sensor clean. 
#define PERFORMCLEANNOW 1
if (PERFORMCLEANNOW) {
    // clean now
    if (sps30.clean() == true)
      Serial.println (DEBUG("fan-cleaning manually started"));
    else
      Serial.println (DEBUG("Could NOT manually start fan-cleaning"));

roberbike, 2021-04-26:
I forgot mention, that during the autoclean periode, there is not particle measurement .

roberbike, 2021-04-26:
/* define new AUTO Clean interval  * Will be remembered after power off * * default is 604800 seconds
 * 0 = disable Auto Clean
 * -1 = do not change current setting 
 * */

#define AUTOCLEANINTERVAL 1

/* Perform a clean NOW ?
 *  1 = yes
 *  0 = NO */

#define PERFORMCLEANNOW 1
  // do Auto Clean interval
  SetAutoClean();

  // start measurement
  if (sps30.start() == true) Serial.println(F("Measurement started"));
  else Errorloop((char *) "Could NOT start measurement", 0);

  // clean now requested
  if (PERFORMCLEANNOW) {
    // clean now
    if (sps30.clean() == true)
      Serial.println(F("fan-cleaning manually started"));
    else
      Serial.println(F("Could NOT manually start fan-cleaning"));

/**
 * @brief: Get & Set new Auto Clean Interval
 *
 */
void SetAutoClean()
{
  uint32_t interval;
  uint8_t ret;

  // try to get interval
  ret = sps30.GetAutoCleanInt(&interval);
  if (ret == ERR_OK) {
    Serial.print(F("Current Auto Clean interval: "));
    Serial.print(interval);
    Serial.println(F(" seconds"));
  }
  else
    ErrtoMess((char *) "could not get clean interval.", ret);

  // only if requested
  if (AUTOCLEANINTERVAL == -1) {
    Serial.println(F("No Auto Clean interval change requested."));
    return;
  }

  // try to set interval
  interval = AUTOCLEANINTERVAL;
  ret = sps30.SetAutoCleanInt(interval);
  if (ret == ERR_OK) {
    Serial.print(F("Auto Clean interval now set : "));
    Serial.print(interval);
    Serial.println(F(" seconds"));
  }
  else
      ErrtoMess((char *) "could not set clean interval.", ret);

  // try to get interval
  ret = sps30.GetAutoCleanInt(&interval);
  if (ret == ERR_OK) {
    Serial.print(F("Current Auto Clean interval: "));
    Serial.print(interval);
    Serial.println(F(" seconds"));
  }
  else
    ErrtoMess((char *) "could not get clean interval.", ret);

void loop() {
  
  uint8_t ret;
  
  // read all ddata
  read_all();
  
  Serial.println("Entering sleep-mode");

  // put the SPS30 to sleep 
  ret = sps30.sleep();

  if (ret != ERR_OK) {
    ErrtoMess((char *) "ERROR: Could not set SPS30 to sleep. ", ret);
  }

  // wait 10000mS = 10 sec
  delay(10000);

  Serial.println("Perform wakeup");
  
  // wakeup SPS30
  ret = sps30.wakeup();

  if (ret != ERR_OK) {
    ErrtoMess((char *) "ERROR: Could not wakeup SPS30. ", ret);
  }

  // as the SPS30 was in measurement mode before sleep that is restored
  Serial.println("measurement mode");
    
  // give time for new air to flow in
  delay(5000);
}

hpsaturn, 2022-06-13:
I'm going to close this issue because is better handling the power management and consumption on firmware level

closed 2022-06-13 by hpsaturn