kike-canaries/canairio_sensorlib #70: CM1106SL-NS and Senseair S11 (sunrise) by i2c low power

roberbike, 2021-04-25:
In order to get very low power consumption battery powered mobile devices, it might be a good idea to add the senseair S11 co2 meter.
Response times are very fast coming out of deepsleep mode.
Providing i2c support could be easier.
For co2 devices without PM sensors, the power supply is 3,3V, no steep-up is needed.

hpsaturn, 2021-04-25:
Good idea @roberbike, of course we need it, but I don't have this sensor for run test, I will need maybe some help from you or do you want make a pull request for that? Anyway I could be help you with that.

roberbike, 2021-04-26:
I have both of them and I  do all the test that you need. My knowledge of programming is scarce so I need help.
I will put here the commands and you can guide me to put them in the right place in the firmware.
Of course,I will compile it and test it .

The last weekend I lost a lot of time trying to test the sps30 with i2c port, without much luck.

hpsaturn, 2021-04-28:
Ok, no problem, thanks @roberbike, when I finish the currrent TTGO development, I will back to sensorlib, maybe the next week.

roberbike, 2021-05-02:
I found this libs for sunrise and cubic sensor with i2c implementation:
https://github.com/albrdev/Sunrise.git
https://github.com/neosarchizo/cm1106_i2c.git

roberbike, 2021-05-02:
Senseair sunrise activation , sleep and wake up mode:

void setup(void)
{
    delay(2500UL);
    Serial.begin(9600);
    while(!Serial);

    Serial.println("Initializing...");
    sunrise.Begin(PIN_EN);
    sunrise.Awake();

    Serial.println("Done");
    Serial.println();
    Serial.flush();

    printStaticInfo();
    printMeterControl();
    printMeasurementSettings();

    sunrise.Sleep();
}

hpsaturn, 2022-06-13:
I'm going to close this issue because is better handling the power management and consumption on firmware level

closed 2022-06-13 by hpsaturn