kike-canaries/canairio_sensorlib #135: CM1106 UART/cm1106_uart.cpp fails to compile.

errolt, 2022-05-03:
## Overview
I'm having issues compiling the lib. It fails on "CM1106 UART/cm1106_uart.cpp". I know how to fix the issue, time.h should be imported in the CM1106 lib, but isn't. It gets annoying fixing it every-time the CM1106 lib is downloaded afresh. How does everyone else work around this issue? Do you fix the downloaded lib? And when you switch to a new target, do you modify the freshly downloaded lib again?

## Steps to Reproduce
In a clean folder, clone the CanAirIO source. Build. All the libs are downloaded, including broken ones.

## Logs
.pio/libdeps/esp32/CM1106 UART/src/cm1106_uart.cpp: In member function 'uint8_t CM1106_UART::serial_read_bytes(uint8_t, int)':
.pio/libdeps/esp32/CM1106 UART/src/cm1106_uart.cpp:390:5: error: 'time' was not declared in this scope
     time(&start_t); end_t = start_t;
     ^~~~
.pio/libdeps/esp32/CM1106 UART/src/cm1106_uart.cpp:390:5: note: suggested alternative: 'tm'
     time(&start_t); end_t = start_t;
     ^~~~
     tm
.pio/libdeps/esp32/CM1106 UART/src/cm1106_uart.cpp:398:17: error: 'difftime' was not declared in this scope
         while ((difftime(end_t, start_t) <= timeout_seconds) && !readed) {
                 ^~~~~~~~

hpsaturn, 2022-05-03:
Thanks for your feedback, it's true, something change, maybe the esp32 base framework and it fails. Let me review the issue to try the find any solution

errolt, 2022-05-03:
Thank you. Really appreciated.

I don't know if I should raise it at the origin, but I'm new to PlatformIO and it looks like it can take a LONG time for fixes to make it into the PlatformIO library store?

But looking at that code, time.h is definitely required...

jcomas, 2022-05-03:
Fixed the compilation in https://github.com/jcomas/CM1106_UART repository. Thanks @errolt for the notice.

hpsaturn, 2022-05-03:
Thanks @jcomas for this fast fix. Please @errolt try again.

errolt, 2022-05-03:
Thank you @jcomas and @hpsaturn! It compiles cleanly!

closed 2022-05-03 by errolt