kike-canaries/canairio_sensorlib #147: Ikea vindriktning-air-quality

markusvankempen, 2022-08-14:
## I have been working with Ikea vindriktning senor using tasmota and adding a wemos +bme680 . I think it it would be a great addition to the sensor lib to add the cheap ikea pm2.5 sensor
https://www.ikea.com/ca/en/p/vindriktning-air-quality-sensor-60515911/
here a example tuturial
https://blakadder.com/vindriktning-tasmota/

hpsaturn, 2022-08-15:
We need first know what library they are using, and test it. Maybe you can send me a simple Arduino hello world running over your stuff? then I could include it

markusvankempen, 2022-08-17:
Hello @hpsaturn take a look at  https://github.com/Hypfer/esp8266-vindriktning-particle-sensor

output example:
11:14:35.250 -> Receiving:....................Done.
11:14:35.547 -> Received PM 2.5 reading: 26
11:14:35.547 -> Current measurements: 25, 25, 26, 25, 25
11:14:37.548 -> We have a PM2.5 state

here a simple code:
/*
  Ikea -  vindriktning
*/
#include "SerialCom.h"
#include "Types.h"

particleSensorState_t state;

// the setup function runs once when you press reset or power the board
void setup() {
  
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);

      Serial.begin(115200);
    SerialCom::setup();

    Serial.println("\n");
    Serial.println("Hello from  vindriktning-particle-sensor");
    Serial.printf("Core Version: %s\n", ESP.getCoreVersion().c_str());
    Serial.printf("Boot Version: %u\n", ESP.getBootVersion());
    Serial.printf("Boot Mode: %u\n", ESP.getBootMode());
    Serial.printf("CPU Frequency: %u MHz\n", ESP.getCpuFreqMHz());
    Serial.printf("Reset reason: %s\n", ESP.getResetReason().c_str());

    delay(3000);
    
}

// the loop function runs over and over again forever
void loop() {
  SerialCom::handleUart(state);
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second

  if (state.valid) {
            printf("We have a PM2.5 state\n");
            Serial.print(state.avgPM25);
            printf("\n");
  }else{
           printf("No PM2.5 state\n");
    }
            
}

hpsaturn, 2022-08-17:
Good. Sound likes that it will be easy. One question. What stuff can you use? TTGO T-Display? My question is for understand the possible pins for a initial test

hpsaturn, 2022-08-21:
good news! @roberbike from our community in Telegram, he test it and it works fine! Thanks to both of you for your feedback and tests. We will be a new version soon:

[image]
https://t.me/canairio/25795

hpsaturn, 2022-09-01:
I'm going to close this issue. The [last version of CanAirIO firmware](https://github.com/kike-canaries/canairio_firmware/releases) already has this support. Thanks!

closed 2022-09-01 by hpsaturn