kike-canaries/canairio_sensorlib #154: Unssuported NOx Sensor request

Geelhem, 2022-10-21:
Could this sensor added to the list of supported sensors 

https://www.mouser.co.uk/ProductDetail/Sensirion/SGP41-D-R4?qs=A6eO%252BMLsxmRqwfOE5NYPVA%3D%3D 

Data Sheet https://www.mouser.co.uk/datasheet/2/682/Sensirion_Gas_Sensors_Datasheet_SGP41-2604356.pdf

Geelhem, 2023-03-02:
Hi any update on this ?

Geelhem, 2023-09-19:
seems like a good alternative way, only is prob is the price! thanks for looking into it.

Geelhem, 2023-10-18:
I have purchased the DFRobot N02 sensor SKU:SEN0471, but it does not seem to be recognised at boot 
[image] Do I need to move the dipswitch for I2C ?

hpsaturn, 2023-10-22:
please test the last version in our OTA dev channel. @roberbike pushed some improvements for this sensor.

closed 2026-09-01 by hpsaturn