kike-canaries/canairio_sensorlib #191: Enhancing Compatibility: Proposing EspSoftwareSerial for Improved Flexibility on ESP32 LILYGO® LoRa32 V2.1_1.6

NatanaelJose, 2024-01-17:
## Summary
Hello,

I'm reaching out as a user of ESP32 LILYGO® LoRa32 V2.1_1.6, and I've observed limitations with the Softwareserial and HardwareSerial libraries. The Canairo library mandates the use of either software or hardware serial, rendering it impractical for use on these boards.

I propose considering the implementation of the EspSoftwareSerial library as a valuable alternative. This would offer users a more flexible and comprehensive solution, allowing for enhanced compatibility and ease of use.

**Implementation Suggestion:**
1. **Add support for the EspSoftwareSerial library in the installation instructions or in the project's README. Additionally, provide a usage example to guide users in implementing EspSoftwareSerial in their projects.**

2. **Usage Example in the case of Sense Air S8:**
   ```cpp
   #include "s8_uart.h"
   #include <SoftwareSerial.h> //EspSoftwareSerial include

   EspSoftwareSerial::UART S8_serial;
   S8_UART *sensor_S8;
   S8_sensor sensor;

   void setup() {
     Serial.begin(9600);
     while (!Serial) {
     }
     // Initialize S8 sensor
     S8_serial.begin(S8_BAUDRATE, EspSoftwareSerial::SWSERIAL_8N1, 21, 22);
     sensor_S8 = new S8_UART(S8_serial);
   }

   void loop() {
     sensor.co2 = sensor_S8->get_co2();
     Serial.print("CO2 value = ");
     Serial.print(sensor.co2);
     Serial.println(" ppm");
     delay(1000);
   }
Thank you for your attention to this matter.

Best regards,
Natanael José

hpsaturn, 2024-01-21:
Thanks for your comment, and sorry for the delay. Well, I think with the last version of CanAirIO Firmware or only using CanAirIO SensorLib is possible to use a different UART or secondary UART passing the pins via the constructor of the la library, like is mentioned in the examples and documentation. How many UARTs do you need for sensors?

hpsaturn, 2025-03-04:
The last firmware via CLI you are able to set a custom UART pins.

closed 2025-03-04 by hpsaturn