kike-canaries/canairio_sensorlib #203: Possible issue with temperature offset logic (and may be with altitude)

melkati, 2024-02-19:
I am very confused with the temperature offsets. I think we have a problem with the logic, in some sensors...

I'm not sure what's going on yet, but something's not right... probably it's a mix of issues...

In my preliminary research I found (unconfirmed, more research is needed) some things to check or fix:

- [ ] On "Sensor Init" (as CO2scdXXInit()) we are **always** calling the function to set an offset. We must probably first get the offset stored on the sensors EEPROM and if it's the same don't set the offset again (maybe this is the reason at each start sensors are taking a while to stabilize the temperature).
- [ ] On sensor init we are calling the function to set or read the offset (and probably the Altitude, but this is another history) before registering the sensor on the Sensorslib so it just errors: CO2scdXXInit() first get the offsets at the sensors EEPROM then compare with toffset (internal variable used by Sensorlib to keep the actual offset) and if necessary, set the new offset but as the sensors aren't registered on Sensorslib these calls just fail.
- [ ] Looks like SCD30 stores the offsets automatically after calling their set_offset function and keeps it between power cicles, but SCD4x doesn't keep the data and a call to the "Persist Settings" function is needed so it's lost after power cycling.
- [ ] On the SCD4x sensors the EEPOM is guaranteed to last for 2000 write cycles. So, we must be very cautious to decide if the "Persist Settings" function should be called from the library when it's needed (more control provided by the library) or from the program using the library (prone to implementation errors but moving the responsibility to the main program's developer).
- [ ] We have not implemented methods Sensors::getTempOffset() so we so that you don't have to call sensors.scd30.getTemperatureOffset() or equivalent directly. 
- [ ] The toffset variable is like a "free electron", it causes confusion and there are temptations to access it directly. It would be necessary to decide and, if necessary, make it private and implement methods to read and modify it in a controlled way.
- [ ] There is an inconsistency for sensors other than SCD30, SCD4x and Sen5X as Sensorlib just ignores offsets for other sensors. Maybe Sensorlib must do a simple subtraction or addition for other sensors to be coherent and give broad compatibility and abstraction to all of them).

Finally, I would like to point out that I want to implement support for low power sensors, starting with SCD41 (which is already "half supported" and is getting a lot of traction) and will have to be taken into account to make it easier later on.

I would like to invite @hpsaturn, @roberbike and @Coscolin to the discussion.

melkati, 2024-02-19:
I moved the sensor registration up in `Sensors::CO2scd30Init()` and `Sensors::CO2scd4xInit()` to fix (2).

I don't see any side effects. Tested with both sensors in CO2 Gadget and looks like it's working fine.

https://github.com/melkati/canairio_sensorlib/commit/42329340867bc088a6653f97509f18c757cef30d
https://github.com/melkati/canairio_sensorlib/commit/6da72010a3cd7a37673c4d7907ed23f48213b38c

Maybe we should do the same for `Sensors::sen5xInit()` but I don't have one to test.

I'll send PR as soon as we have a new branch to work on.

melkati, 2024-02-20:
I just sent a PR to fix last commit and (5). https://github.com/kike-canaries/canairio_sensorlib/pull/204

I didn't add sen5x to Sensors::getTempOffset() and I don't know how it works and don't have one to test.

hpsaturn, 2024-02-20:
Maybe @roberbike knows about this sensor.

BRKMK, 2024-02-20:
Yes, I have one sen5x from an ikea device. I will test it in a few days.

hpsaturn, 2024-02-21:
Thanks @melkati 
On my side, right now I don't have SCD30 here to perform tests. Only I have a new SCD40 builtin sensor, included in the AirQ unit of M5. I'm going to check if is possible replicate the issue there. On the other hand, I could help in the review code of any advance. Thanks again.

melkati, 2024-02-21:
I did the same tests with an SCD40 sensor and the same thing happened.

One tip: the programmer using the library must set the float variable `sensors.toffset`, **prior to `sensors.init()`,** to the same temperature offset the program using the library has stored as "actual offset temperature" for everything to be "on sync".

Maybe we should create a method to set this variable **before calling `sensors.init()`** so programmers do not have to access the variable directly.

I can't find any problem with manually setting the variable, before calling `sensors.init()`. @hpsaturn, what do you think?

melkati, 2024-02-21:
> Yes, I agree. It is normal in some libraries. If we need that, please do it.

Is initTOffset a good name?

hpsaturn, 2024-02-25:
Yes, these names are good. I guess that we should admit the two possibilities, positives and negatives.

melkati, 2024-02-27:
Just sent PR with these two methods. I didn't make tOffset variable private to maintain backward compatibility. Maybe we should do it.

melkati, 2024-03-01:
On my side, this PR is ready for merging.

CO2 Gadget is using this branch (in my fork) and nobody reported issues with sensors, so I think it's working fine.

Point 7. is still pending to improve in the future. I tried to open it on their own issue, so it's not forgotten, but I receive an error:

[image]

hpsaturn, 2024-04-21:
Thanks @melkati for your contributions and time. Then I'm going to close this issue. Thanks!

closed 2024-04-21 by hpsaturn